tinypico/tinypico-arduino #7: Slow Boot times

CrispinP, 2020-02-18:
Hi,

I have an off situation here with your board running the sketch below.

You board:
[image]
[image]

DevKit ESP32
[image]

Sketch
```
void setup() {
  Serial.begin(115200);
  Serial.print("I've been alive for ");
  Serial.print(millis());
  Serial.println("ms");
  
  // put your setup code here, to run once:

}

void loop() {
  // put your main code here, to run repeatedly:

}
```

I'm building a project which deep-sleeps, counts pulses with the ULP, wakes up after a set amount, logs on to wifi, publishes the counter with mqtt and goes back to sleep.
The normal runtime on the devkit is around 300ms including wifi.

> 
> [33 ] ULP wakeup, saving pulse count
> [33 ] Pulse count from ULP:4
> [33 ] Wifi Setup - Starting
> [68 ] Wifi Setup - Setting up WiFi Manager.....
> [69 ] Wifi Setup - Setting up WiFi Manager.....done
> [69 ] Got stored Credentials. Timeout 60s
> [70 ] Wifi Setup - Done
> [72 ] Wifi Setup - Connecting
> [75 ] Connecting to CrappyIoTconnected! Local IP: 337512970
> [259 ] Setting up MQTT Starting
> [259 ] Attempting MQTT connection...
> [292 ] MQTT connected!
> [292 ] Leaving MQTT connect
> [292 ] Setting up MQTT Complete
> [296 ] Disconnecting from MQTT.
> [297 ] MQTT disconnected.
> [297 ] Finished. Going to sleep
> [297 ] Entering deep sleep

You can see that being alive and "useless" for 1500ms is a battery killer in this way :)

Why would it be taking this long before it actually hit the setup?

C

UnexpectedMaker, 2020-02-18:
Do you have PSRAM enabled? If so, it does a RAM check on boot. Disable the PSRAM and see if that helps. Other than the extra RAM, there is no difference between a TinyPICO and any other ESP32 in terms of Booting.

CrispinP, 2020-02-19:
Ah yes. Found the option to disable PSRAM. Arduino IDE > Tools > PSRAM > Enable / Disabled 

It now boots in 33ms :)

Thanks!

closed 2020-02-19 by CrispinP