tinypico/tinypico-arduino #8: boards.txt may define the wrong pins_arduino.h file for TinyPICO

Paraphraser, 2020-03-19:
I think I have found a problem in the TinyPICO setup. This may not be the correct place to report it but it seems better than nothing.

I'm using:

* macOS 10.14.6
* Arduino 1.8.12
* esp32 1.0.4
* TinyPicoHelperLibrary 1.3.0

The Arduino IDE is set to:

* Tools » Board » ESP32 Arduino » TinyPICO

Consider this sketch:

```
#include <Arduino.h>
#include <TinyPICO.h>

void check (const char * tag, uint8_t value, uint8_t expectedValue) {

    if (value != expectedValue) {

        Serial.printf(
        	"Symbol %s has value %d, expectedValue \n",
        	tag,
        	value,
        	expectedValue
        );
        
    }
    
}

void setup() {
    
    Serial.begin(115200); while (!Serial); Serial.println();
    
    /* these symbols should be available from pins_arduino.h */
//    check("APA_POWER",APA_POWER,13);
//    check("APA_DATA",APA_DATA,2);
//    check("APA_CLK",APA_CLK,12);

    /* these symbols should be available from TinyPICO.h */
    check("DOTSTAR_PWR",DOTSTAR_PWR,13);
    check("DOTSTAR_DATA",DOTSTAR_DATA,2);
    check("DOTSTAR_CLK",DOTSTAR_CLK,12);

    Serial.println("Complete");

}

void loop() { }
```

That compiles OK. But if I uncomment **any** of these three lines:

```
//    check("APA_POWER",APA_POWER,13);
//    check("APA_DATA",APA_DATA,2);
//    check("APA_CLK",APA_CLK,12);
```

then the sketch won't compile, giving errors like:

```
sketch_tinyPicoPins:66:21: error: 'APA_CLK' was not declared in this scope
     check("APA_CLK",APA_CLK,12);
                     ^
```

The path:

```
~/Library/Arduino15/packages/esp32/hardware/esp32/1.0.4/variants/tinypico/pins_arduino.h
```

contains these lines:

```
static const uint8_t APA_POWER = 13;
static const uint8_t APA_DATA = 2;
static const uint8_t APA_CLK = 12;
```

so, all other things being equal, those symbols should be defined and available to the sketch.

Backtracking to:

```
~/Library/Arduino15/packages/esp32/hardware/esp32/1.0.4/boards.txt
```

finds this line:

```
tinypico.build.variant=pico32
```

and I *think* that means that, instead of the path to pins_arduino.h given above, the compiler is being directed to:

```
~/Library/Arduino15/packages/esp32/hardware/esp32/1.0.4/variants/pico32/pins_arduino.h
```

Comparing the two pins_arduino.h files produces:

```
$ diff ~/Library/Arduino15/packages/esp32/hardware/esp32/1.0.4/variants/pico32/pins_arduino.h ~/Library/Arduino15/packages/esp32/hardware/esp32/1.0.4/variants/tinypico/pins_arduino.h 
55a56,59
> static const uint8_t APA_POWER = 13;
> static const uint8_t APA_DATA = 2;
> static const uint8_t APA_CLK = 12;
> 
```

which explains the observed behaviour.

UnexpectedMaker, 2020-03-19:
This has been fixed and has been waiting on the release of the 1.0.5 ESP32 Arduino update from Espressive for a looooong while ;)

That's why I have the extra declarations in the TinyPICO helper, so you can avoid using the "Arduino pins" ones.

V 1.0.4 was released Oct 2nd. It's been a long wait for 1.0.5, but lots of world issues have gotten in the way :(

Paraphraser, 2020-03-19:
Okey dokey but could I suggest leaving this issue open until then in case anyone else is wandering down the same path.