tinypico/tinypico-arduino #11: Does deep sleep work on Arduino?

brownedon, 2020-11-05:
Tried the DeepSleep example from the Arduino examples, stays at 20mA, with no decrease, sleep or not.  I can do this on the ESP32 DOIT, without issue, sleep goes down to a couple mA.

brownedon, 2020-11-28:
Ok, Deep Sleep doesn't work AT ALL on a Lipo.  Works on USB.  If I unplug the USB while it's in sleep, I see the current usage is below 1ma.  When the timer wakes up current goes to 23, and stays there.  Looks like the device doesn't reset again.

UnexpectedMaker, 2020-11-28:
What version of the ESP32 core are you using? and how are you taking it into deep sleep?
Deepsleep def works and works fine on battery.

brownedon, 2020-11-28:
Looks like v3.2.3-14-gd3e562907.

I’m using Examples->Deep Sleep->Timer Wakeup.  I see no reduction in current consumption versus a DOIT.

If I unplug it from USB when it’s in sleep, I can see the reduced current consumption, then it comes out of sleep, current maxes and then it doesn’t go back into sleep again.  Basically it never restarts on battery.

UnexpectedMaker, 2020-11-28:
That's not an ESP32 Arduino Core version. Latest is 1.0.4 installed from the Arduino IDE. 
There is also 1.0.5rc2 available from github
https://github.com/espressif/arduino-esp32/tags
 
How are you installing the ESP32 Arduino Core? 
How the TinyPICO is powered will have no influence on how deepsleep operates or wakes.

brownedon, 2020-11-28:
Right, I have 1.04.  I also tried the other versions last night.  I’m installing from the board manager.

 

https://github.com/espressif/arduino-esp32/tags

UnexpectedMaker, 2020-11-28:
Can we move this to my discord server? posting back and forwards on GitHub is slow and painful.
https://discord.com/invite/xAHpApP

Jump into the #arduino help channel and we can discuss more easily.
Cheers

brownedon, 2020-11-29:
I've got it working now.  Was measuring current on the wrong side, gotta do it across negative not positive.

closed 2020-11-29 by brownedon