tinypico/tinypico-arduino #12: IsChargingBattery() doesn't work

sorgelig, 2021-09-17:
It returns TRUE when USB is not connected.
Probably because no one pulls up when no USB is connected?

UnexpectedMaker, 2021-09-17:
It works correctly. It's designed to tell you if the battery is charging or not, not to tell you if USB is connected.
You cannot put a pull up on STAT - it interferes with the operation of the PMIC.

sorgelig, 2021-09-17:
I don't need to check if USB connected.
I just used this function to tell if battery is changing. And it still shows charging even if i unplug the USB.
What i'm doing wrong?
```
tp.DotStar_SetPixelColor(tp.IsChargingBattery() ? 255: 0, 255, 0);
```

UnexpectedMaker, 2021-09-17:
You are not doing anything wrong, but the PMIC cannot tell the difference between no battery and a fully charged battery, so it will not understand the charge stat when no battery is present. It's why the LED flickers on and off.

So, you will not get reliable results when no battery is present. As I said, it is ONLY for checking the charge state when a battery is present, not when one is not present.

What revision TinyPICO do you have? V1 or V2? Most V2 boards have a VBUS sense on IO9, so you can first detect is USB is present. if it's not present, don't use tp.IsChargingBattery() and just set the LED to "not charging" - if it is, you can then use tp.IsChargingBattery() to get the charge state.

sorgelig, 2021-09-18:
revision of board is V2C5 - i can't find schematics of this exact revision. Does it has VBUS connected to GPIO9?

> So, you will not get reliable results when no battery is present. As I said, it is ONLY for checking the charge state when a battery is present, not when one is not present.

obviously battery is connected, otherwise nothing will power the board. LiOn Battery is connected, so i simply connect/disconnect USB to see the difference between USB and battery powering. In both cases i get "charging" result. Btw, when no battery is connected then charge LED is blinking if USB is connected.

sorgelig, 2021-09-18:
modified TinyPICO.cpp:
```
#define VBUS_SENSE 9

TinyPICO::TinyPICO()
{
    ...
    pinMode( VBUS_SENSE, INPUT );
    ...
}

bool TinyPICO::IsChargingBattery()
{
    if(!digitalRead( VBUS_SENSE )) return 0;
    ...
}
```

now it reports charging correctly.
Should it be added to helper?

UnexpectedMaker, 2021-09-18:
> modified TinyPICO.cpp:
> 
> ```
> #define VBUS_SENSE 9
> 
> TinyPICO::TinyPICO()
> {
>     pinMode( VBUS_SENSE, INPUT );
>     ...
> }
> 
> bool TinyPICO::IsChargingBattery()
> {
>     if(!digitalRead( VBUS_SENSE )) return 0;
>     ...
> }
> ```
> 
> now it reports charging correctly.

Excellet!

> Should it be added to helper?

No, as not all TinyPICOs have VBUS sensing, so the code will fail if it doesn't have it and there's no way to detect from querying the TP if the functionality is there or not.

sorgelig, 2021-09-18:
git it.

closed 2021-09-18 by sorgelig